BerkeleyHCI/PolymorphicBlocks #22: Compiler core rewrite

ducky64, 2020-12-24:
Some of the planned changes are pretty fundamental, in that it's likely the compiler core will need to be completely rewritten anyways. Current plan is to rewrite the core in Scala, which should make the code a bit more cleaner.

Planned changes:
- #20 link arrays
- #13 directed constraints
- Possible directed generators? As an alternative to the current subset tightening model, directed generators would "fire" when its inputs are ready. More thought would need to be put into supporting fixed-point design practices / loops
- Possibly #9, improved metadata fields
- Would need a Scala <-> Python interface to call into Python generators and still allow a Python top-level object

Design for these performance improvements, by having a persistent cache object, ideally one that can be serialized
- #17 cached generators
- #7 cached libraries
- #18 might be obsoleted, though there might be places to eliminate CopyFrom in the frontend (but likely to be less important, since the frontend execution is pretty shallow)

### Compiler structure
TODO

rohit507, 2020-12-28:
> Some of the planned changes are pretty fundamental, in that it's likely the compiler core will need to be completely rewritten anyways. Current plan is to rewrite the core in Scala, which should make the code a bit more cleaner.

Agreed, seems like a generally good plan. 

>     * Possible directed generators? As an alternative to the current subset tightening model, directed generators would "fire" when its inputs are ready. More thought would need to be put into supporting fixed-point design practices / loops

So, if we assume that generators will trigger in some total order<sup>[1]</sup> and we don't need to have semantics for generators firing "simultaneously"<sup>[2]</sup>, then we shouldn't have much issue in the short term. The problem is in the longer term we either need to have non-deterministic generators<sup>[3]</sup> (very hard for us, probably requires us to craft a custom (non-E)DSL) or require that generators meet a bunch of totality criteria that makes them much harder for their authors to make. 

It might be worth considering if we could do anything with repeated evaluation of a generator. Generators have "types" that determine when we should fire them and are supposed to work for *every* input that's a subset of that type. That should mean we could fire generator `A`, do a bunch of other generation and propagation (that should only refine the input, never make it so the type of the original part doesn't fit the generator), delete all the stuff `A` added, and run `A` again. This should mean that `A` has a chance to generate a better implementation given the new information from the rest of the design, even though that new information was prompted by the first "draft" design that `A` generated. 

Not sure if that would work in practice and it assumes that generators are mostly deterministic but it might allow for human usable fixed point-ish semantics that produce better optimized designs. 

<sub>**[1]**: I think this is a fine assumption if we're willing to forgo some sort of logical determinism<sup>[4]</sup>, and that we shouldn't really be trying to have that kind of determinism with this version of the system.</sub>

<sub>**[2]**: As opposed to what I assume you're doing when generators `A` and `B` can both fire at the same time. Namely, the compiler would pick one of the generators, say `A`, evaluate it, propagate the updates as needed, and then fire `B`. Doing something like "fire `A`, fire `B`, propagate" would be a better match for the fixed point style of evaluation but would make it harder to write generators that function well and cause more unnecessary refinement failures.</sub>

<sub>**[3]**: In the sense of non-deterministic algorithms in complexity theory. Namely that all possible paths of evaluation occur simultaneously and that the results of all the paths are collected together. Not non-deterministic in the sense that you're sampling from all possible evaluations.</sub> 

<sub>**[4]**: By logical determinism, which isn't a real term, I mean that the output of the compiler should be deterministic given only the input design and generators without any hacky tricks to ensure a consistent evaluation order. You could say "Always run the first available generator when sorted by alphabetical order" or something and have determinism, but now that determinism is decoupled from the logical formalism that backs the compilation process (Not that said formalism is well defined right now, but you get the point). This also applies to having a deterministic order for firing propagators instead of propagator firing following a diamond law.</sub> 

>     * Would need a Scala <-> Python interface to call into Python generators and still allow a Python top-level object

Between `pickle` and `inspect` this should be possible, no idea how difficult though. 

> ### Compiler structure

Mmm. 

>     * A directed graph data structure for parameters, parsed from the directed constraints. In addition to being useful for optimized solving, this could also be used as part of a constraint tracing visualization for debugging

Hmm, you're only handling directed constraints in this compiler?

>  [algorithm sketch ...] 

Seems broadly good.

>           * This can be implemented as a fixed point algorithm (naive simple version - inefficient) or using a directed graph that keeps track of unevaluated directed assignments (edges pointing into (inputs/ RHS) and out of (outputs / LHS) each constraint, with the evaluation attempts triggered by evaluation of RHSs and on instantiation).

Be warned, even the directed model can be massively inefficient if there's any cycles. In general you're going to want some sort of dependency order analysis to choose which edges to run. (Skipping this is what made my second attempt at a propagator system fail.) 

There's a bunch of literature on efficient evaluation order for logic programming languages and complex database queries that ends up being really relevant to this. Also lookup the Rete algorithm and its more recent derivatives for more insight into efficient rule evaluation. 

>           * Unassigned parameters would be an error (everything must have a default in the frontend -

Is that a good idea? Weren't unassigned parameters going to create errors for the user that signaled under-constrained designs? 

>           * Existing "==" constraints would be treated as assertions rather than assignments

skeptical-face.jpg 

[... cut ...]  (and arguably even if inputs are the same). 

>        * Errors should be batched (like type checkers) instead of failing first error. Source locators should be reported where possible (eg, failing assertion line, failing block instantiation line). More ambitiously, constraints should indicate the source locator of the error, though this is likely to be difficult to do well because constraints may have many inputs

Agreed, python is surprisingly capable of helping with this thanks to `inspect`.

ducky64, 2020-12-29:
**Generators and run order**: so combined with that parameters can't be overassigned, when a generator is ready, the inputs should be in their final state, so order shouldn't matter (but practically, if multiple generators were available, the one that runs would be arbitrary; at the end everything should converge to the same result).

I've thought about ideas to handle multiple runs to tighten bounds, but I think I'd like concrete examples to drive that. The idea I'm leaning towards is extending the Range tightening feature we have now to allow that to re-trigger generators, and iterating until a fixed-point solution or some number of iterations. Though, not completely sure how the control and specification of these should be handled.

**Compiler structure**: Yes, only planning to handle directed constraints for propagation. Undirected constraints would be treated as assertions. Also, there shouldn't be any cycles in this model (cycles would be detected and cause an error), in theory this should be a relatively simple DAG traversal? Though I'll take a look at the Rete algorithm

**Unassigned parameters**: Sorry, to clarify a bit: unassigned parameters can happen and will error to signal an under-constrained design. But I expect many parameters will have sensible defaults, eg voltage input ports would have a default voltage limit of (-inf, inf). Future analyses can detect propagation of default values, eg to warn on use.

**== as assertions**: The alternative would be to stylistically match `(var) == (expr)` constraints as directed assignments, but the disadvantages would be that it's not obvious that it's an assignment, and it's not obvious it's directed. I think I'd prefer a obviously different construct, maybe `var.assign(expr)`.

**Cache**: The idea is to cache a generator based on its input values. Some generators are expected to be expensive to run (eg, part table search), so it makes sense to save the result of the first run, and some generators get run a lot even within the same design (eg, 0.1uF capacitors). Generators would be explicitly marked as cache-able.

The other cache item worth thinking about in the future would be to cache entire sub-trees, though mainly for re-compilation speed. For example, if a buck converter is instantiated with the same inputs, in theory we can avoid elaborating the sub-tree from scratch and drop the cached one in. But this is harder than the generator case, where inputs are explicitly marked and hence very bounded. It's also less clear how much time this would save, since I expect library instantiation to be relatively fast,
because the number of parameters are limited (order tens?) and it's in JVM instead of Python.

PR #32 (2021-04-17): Core refactor and Scala compiler
An absolutely massive and major refactor, and with some things still broken.

Major changes:
- Eliminate the Python-based compiler, and instead reimplemented it in Scala. (resolves #22)
  - The choice of Scala is a mix between a high-level and expressive language while also being high performance (at least much higher than Python)
  - The new compiler is based on expanding the design tree frontier, combined with a dependency graph for parameter propagation and generator invocation.
  - The HDL remains in Python, and they communicate through gRPC. See compiler/r [... cut]
closed 2021-04-17 by ducky64