BerkeleyHCI/PolymorphicBlocks #70: Adding more type information to exprs.

rohit507, 2021-11-07:
As discussed on slack. My ideal way to add further type information would be by adding more ops/opcodes to expr.proto, so that we aren't overloading operations needlessly. 

In my ideal case (mocked up [here](https://gist.github.com/rohit507/d10ac6f98c921d6c3446052389e7e246)) we'd:
  - Add a unary expression that handles non-set unary operations. (lines 153-181)
    - Basic ops like NEGATE, NOT, and INVERT. (157-164)
    - Range value extraction ops like MIN, MAX, CENTER, and WIDTH (166-176)
  - Convert `ReductionExpr` to `SetExpr` (or similar) and use it for all unary operations with sets as input. (183-258) 
    - Convert MINIMUM and MAXIMUM to only apply to sets (217-227)
    - Add pointwise NEGATE and INVERT (246-254)
  - Remove unnecessary binary operations like SUB (43-44) and DIV (49-50), in favor of using the appropriate unary op to find the negation/inverse. 
  - Add BroadcastExpr for operations that apply an operation pointwise to all elements in a set. (138-151) 

This would require some minor changes to how things are encoded but reduced ambiguity and complexity a lot. For example: 
  - `a - b = a + (unaryNegate b)`
  - `1/sum(1/{a,b,c}) = unaryInvert( setSum( setInvert( {a,b,c} )))`
  - `{a,b,c}/3 = broadcastMult( {a,b,c} , unaryInvert(3))`
  - `3/{a,b,c} = broadcastMult( setInvert( {a,b,c} ), 3)`

Thoughts?

rohit507, 2021-11-07:
Updated the gist to make the various type signatures more explicit, also removed comments implying that subset and intersection are applicable to set-like types.

ducky64, 2021-11-08:
The mockup case looks reasonable, but some concerns:
- I can see how ADD / NEGATE and MULT / INVERT could be a more minimalistic encoding while being functionally equivalent. One question is how much we want to design the IR for usability - for example, should users (or rather, what kind of users) be expected to look at IR when debugging, The ADD / NEGATE would be a weirder encoding, though maybe the response to that is "users should only look at the Python code and the IR is a blackbox".
  - This also limits what tools can do with displaying expressions. But maybe the right answer there is instead of trying to pretty-ify IR, just find / link to the relevant line of source Python?
- Does it make sense to replace either LT or GTE, and GT or LTE with the NOT equivalent?
- One downside is that there's a lot more opcodes, even if they all mean similar things (though the benefit is it's more explicit). So a bit more code in the compiler, and also a longer proto definition (which I kind of like to keep minimalistic for readability - I think the proto is a good overview of the overall system)

What's the timeline you'd like on this? My current preference is to push the SMU board through on current infrastructure, then maybe begin a round of internals overhaul.

rohit507, 2021-11-09:
> What's the timeline you'd like on this? My current preference is to push the SMU board through on current infrastructure, then maybe begin a round of internals overhaul.

I'm basically blocking on a decision with this. If we decide to do it there's other work I could be doing for 1-2 weeks while I wait. If not then I really ought to get banging my head against workarounds right away. 

> * I can see how ADD / NEGATE and MULT / INVERT could be a more minimalistic encoding while being functionally equivalent. One question is how much we want to design the IR for usability - for [...]

Yes, but I don't think that having users see an occasional `(-a)+b` or `(1/d)*c` is a big burden. It's slightly unaesthetic but it's perfectly readable. 

I've been assuming that the IR isn't something people will be looking at without tools to help them. As it stands I've has to do a lot of reading the IR for debugging already and it's fundamentally not suited for humans without an intermediate layer. Tools like the block diagram viewer are necessary, and those can simply pattern match on `a+(-b)` or `c*(1/d)` and provide a nicer rendering. It's literally those two cases and no others, not even the ones with the arguments flipped need prettifying as presumable there's some reason for having the flipped term in the lhs. 

>       * This also limits what tools can do with displaying expressions. But maybe the right answer there is instead of trying to pretty-ify IR, just find / link to the relevant line of source Python?

I think this is the correct answer in the longer term, and part of what the metadata field exists to allow experimentation with. 

>     * Does it make sense to replace either LT or GTE, and GT or LTE with the NOT equivalent?

Could go either way on a larger scale but it doesn't create issues for me. Arguably you could drop everything except LTE or GTE and still be able to cobble together every ordering operation but the goal isn't to minimize opcodes for the sake of minimizing opcodes. 

My changes try to minimize how ambiguous the meaning of any particular opcode is. Removing SUB and DIV is incidental to introducing both unary and pointwise versions of NEGATE and INVERT. Otherwise we'd need separate binary and broadcast versions of SUB and DIV in addition to binary and broadcast ADD and MULT. The semantics of those are more complex than the corresponding variants of NEGATE and INVERT. 

> * One downside is that there's a lot more opcodes, even if they all mean similar things (though the benefit is it's more explicit). So a bit more code in the compiler, and also a longer proto [...]

Yeah, the extra compiler code is the big downside but this would be trading logic for pattern matching. Making things more explicit means that there's fewer cases to actually handle. 

As for the protos, I think you're conflating two types of readability. Overloading constructs makes it harder to understand the system by reading the protos, since people need to aware of all the different things an expression might mean at any point. You can't explode things too much, which is why I'm still advocating for ops that can handle both ranges and scalars, but those are constructs that both represent the same thing (a set of values, even if some are singleton sets). 

By contrast, sets in the proto don't actually represent mathematical sets. They're closer to indexed arrays or structs and have a different semantic meaning and aren't used to represent a bunch of possibilities. 

Trying to minimize the syntax used to express ideas will just make you a mathematician. The sort of person who can write a complex concept on a board in 30 seconds using notation tailor made to express that particular idea. Thing is you'd also be creating the students who have to spend half an hour decoding that one line by figuring out exactly what each new tick or squiggle means this time for this particular problem.

ducky64, 2021-11-09:
Ok, I think I'm happy with the trade-offs in your proposed proto. Feel free to PR the proto changes, and I'll work on updating the frontend and compiler for those changes probably next week. I'd expect a week or two of work after that, but it may go faster.