BerkeleyHCI/PolymorphicBlocks #77: Improve component (mostly generators) unit testing

ducky64, 2021-11-16:
So the current state of unit testing for generator blocks is not great - only a few have tests, and the test are quite verbose (which probably contributes to why there's so few of them).

The main problem right now is that to write a test, it needs a bunch of things that are partially redundant and separate:
- Of course, the DUT instantiation
- ... which need to be wrapped in a top-level block - which is basically all boilerplate
- ... and which needs to have its ports connected - which then needs dummy block definitions to provide those ports
- and the test runner code (which is unavoidable, but ideally wouldn't be in a separate block than the DUT instantiation)
- and the check code, which is structured as raw variable gets

## Ideas
Mostly the issue is with the DUT instantiation being in a separate top-level block from the test code, and the need to connect all its ports. Ideas around it:
- **Waived checks + refinements**
  - The main reason we need the dummy blocks is to connect ports so we don't get required-port-not-connected errors. Idea here would be to define errors to ignore, and just let those required port checks fail. Or maybe ignore all errors so there isn't any implicit failure, and all failures would stem from user-written asserts for each test case.
  - Downside is that it's generating effectively broken IR, and there's no guarantee this will continue to work as the compiler changes.
  - False-negatives are more likely.
- **Automatically generated scaffolding**
  - Automatically generate the top-level (wrapper) block and the dummy blocks, given only the instantiation of the DUT.
  - Or for the dummy blocks, define a test-specific registry of port types to their dummy blocks. Maybe also some way to override this, either in the test invocation (choosing a different dummy block for a particular test case), or through the refinement system (choosing different parameters for the dummy block or port).
  - The current IR doesn't have good support for a block that can generate into distinct library elements, so some work may be needed there.
  - Downside is that this requires more infrastructure, but at least it tests a complete IR model.
- **Test core logic only**
  - Since most of the test is generators, that is really the only logic that needs to be tested. This requires the core logic for each generator to be split out so the inputs and outputs are purely values and functional, since assignment constraints would be more difficult to parse.
  - Downside: not end-to-end, requires this re-structuring which makes writing blocks worse.
  - Upside: pretty minimal test, especially one that would be fast since it's not invoking the compiler. Not sure the pros outweigh the cons, though.

Leaning towards automatically generated scaffolding, if it turns out to be feasible.

rohit507, 2021-11-17:
Generating scaffolding seems like the best idea here, especially since core logic tests seem like they're very generator specific. 

Also, it seems like generated scaffolds would be really well complemented by some sort of property testing. 

I think I need a better idea of what the test suite looks like for some example generators. What exactly are you testing for?

ducky64, 2021-11-17:
There are mainly simple spot-checks that the generator is doing the right thing, for example that [the non-inverting amplifier chooses the right resistors](https://github.com/BerkeleyHCI/PolymorphicBlocks/blob/smu/electronics_abstract_parts/test_opamp.py#L49) or the [resistor picker chooses the right parts](https://github.com/BerkeleyHCI/PolymorphicBlocks/blob/master/electronics_lib/test_resistor.py).

A different question is doing more checking of the full examples - currently, the only check is they compile, but maybe it's worth making sure their netlists don't change (or rather, flagging netlist changes to be manually reviewed).