BerkeleyHCI/PolymorphicBlocks #151: Remove array literal

ducky64, 2022-09-08:
This seems redundant with the ValueExpr.Array array-from-arbitrary-expression constructor. The main reason literals are used are to ensure only literals are passed through generators, but those can be modified to take arbitrary ValueExpr and fail on non-literal cases.

rohit507, 2022-09-13:
> This seems redundant with the ValueExpr.Array array-from-arbitrary-expression constructor. The main reason literals are used are to ensure only literals are passed through generators, but those can be modified to take arbitrary ValueExpr and fail on non-literal cases.

The original reason for this was to make the literal message in the proto structurally distinct from expressions. Remove that and you need to add a bunch of checks for whether each expression is a literal wherever you use them. 

This makes more sense in a well-typed language like Scala or Haskell.

I'd treat reusing ValueExpr as a code smell but that might not be worth it if it adds a lot of boilerplate.

ducky64, 2022-09-15:
I think in the end those applications that require literals are requiring a subset of the IR. The current structure with array literals means that the types are well bounded (structurally impossible to have a non-literal value), but even without array literals I don't think the code would look too much different since it just tests cases - it can just fail if it sees a non-literal. Scala can statically check for match completeness including on proto, I don't think Python can do that.

This isn't too bad of an issue with arrays, since it's just one more thing. But if we have more containers / collections in the future we also need to grow the parallel structures maintained, which may be undesirable. This is a possible way to minimize the IR, but I'm not super attached to it.