BerkeleyHCI/PolymorphicBlocks #251: Compiling with blocks defined outside of library.

imuli, 2023-05-08:
After refactoring my board into nice sub-modules, I was very confused when it no longer compiled!

Example following the blinky tutorial:

```python
from edg import *

class Lf21215tmr_Device(FootprintBlock):
  def __init__(self) -> None:
    super().__init__()
    self.vcc = self.Port(
      VoltageSink(voltage_limits=(1.8, 5.5)*Volt, current_draw=(0.5, 2.0)*uAmp))
    self.gnd = self.Port(Ground())

    self.vout = self.Port(DigitalSource.from_supply(
      self.gnd, self.vcc,
      current_limits=(-9, 9)*mAmp,
      output_threshold_offset=(0.2, -0.3)
    ))

  def contents(self) -> None:
    super().contents()
    self.footprint(
      'U', 'Package_TO_SOT_SMD:SOT-23',
      {
        '1': self.vcc,
        '2': self.vout,
        '3': self.gnd,
      },
      mfr='Littelfuse', part='LF21215TMR',
      datasheet='https://www.littelfuse.com/~/media/electronics/datasheets/magnetic_sensors_and_reed_switches/littelfuse_tmr_switch_lf21215tmr_datasheet.pdf.pdf'
    )

class Lf21215tmr(Block):
  def __init__(self) -> None:
    super().__init__()
    self.ic = self.Block(Lf21215tmr_Device())
    self.pwr = self.Export(self.ic.vcc, [Power])
    self.gnd = self.Export(self.ic.gnd, [Common])
    self.out = self.Export(self.ic.vout)

    self.cap = self.Block(DecouplingCapacitor(capacitance=0.1*uFarad(tol=0.2)))
    self.connect(self.cap.pwr, self.pwr)
    self.connect(self.cap.gnd, self.gnd)

  def contents(self) -> None:
    super().contents()

class BlinkyExample(SimpleBoardTop):
  def contents(self) -> None:
    super().contents()
    self.usb = self.Block(UsbCReceptacle())
    self.reg = self.Block(VoltageRegulator(3.3*Volt(tol=0.05)))
    self.connect(self.usb.gnd, self.reg.gnd)
    self.connect(self.usb.pwr, self.reg.pwr_in)

    with self.implicit_connect(
        ImplicitConnect(self.reg.pwr_out, [Power]),
        ImplicitConnect(self.reg.gnd, [Common]),
    ) as imp:
      self.mcu = imp.Block(IoController())
      # if the following two lines are commented out it compiles fine
      self.mag = imp.Block(Lf21215tmr())
      self.connect(self.mcu.gpio.request('mag'), self.mag.out)

      (self.sw, ), _ = self.chain(imp.Block(DigitalSwitch()), self.mcu.gpio.request('sw'))

      self.led = ElementDict[IndicatorLed]()
      for i in range(4):
        (self.led[i], ), _ = self.chain(self.mcu.gpio.request(f'led{i}'), imp.Block(IndicatorLed()))

  def refinements(self) -> Refinements:
    return super().refinements() + Refinements(
    instance_refinements=[
      (['reg'], Tps561201),
      (['mcu'], Esp32_Wroom_32),
    ],
    instance_values=[
      (['mcu', 'pin_assigns'], [
        'led0=26',
        'led1=27',
        'led2=28',
        'led3=29',
       ])
    ])

if __name__ == "__main__":
  compile_board(BlinkyExample, ("./", "blinky"))
```

Upon running `python blinky.py` I get a lot of errors that look like:

```
Traceback (most recent call last):
  File "/nix/store/2lq7wa3rj44nr4jjr8ap44h5y28hcy61-python3-3.10.10-env/lib/python3.10/site-packages/edg_hdl_server/__main__.py", line 100, in run_server
    cls = class_from_library(request.get_library_element.element,
  File "/nix/store/2lq7wa3rj44nr4jjr8ap44h5y28hcy61-python3-3.10.10-env/lib/python3.10/site-packages/edg_hdl_server/__main__.py", line 77, in class_from_library
    cls = getattr(elt_module, elt_split[-1])
AttributeError: module '__main__' has no attribute 'Lf21215tmr'
```

which seems to indicate that it can only pull classes from the `edg` library, not from the user's modules.

imuli, 2023-05-08:
This can be worked around by creating a seperate `__main__.py` like this:

```python
from .blinky import BlinkyExample
from edg import compile_board

if __name__ == "__main__":
  compile_board(BlinkyExample, ("./", "blinky"))
```

Otherwise it appears that `edg_hdl_server` loads itself as `__main__` and tries to find the class inside itself.

ducky64, 2023-05-08:
Yeah, working on fixing this, I think I found the issue. The problem is that the file run in the `python` command is internally referred to as `__main__`, but the compiler spins up a separate HDL interface server which has itself as `__main__`, and it can't resolve `__main__` references in the original file since it has a different idea of what `__main__` is.

I think in general, if you put libraries in the not-`__main__` file it should work, but a fix is coming soon.

ducky64, 2023-05-08:
Should work now, and a new version (0.0.1) should be available on pip.

PR #252 (2023-05-08): Add packaging tests, support blocks defined in __main__
Add local packaging run to workflow tests, with an abbreviated (blinky-only) unittest and top-level test.

Resolves #251, where blocks defined in `__main__` cause an error when the HDL server tries to resolve those blocks, because the HDL server is a separate Python instance and thinks itself is `__main__` (instead of the original file). The fix is to detect when a library's module is `__main__` and effectively guess the correct original module name by using the filename. Might be a better way to do it, but this works, and is largely what [... cut]
commit df4a8264 (2023-05-08): Add packaging tests, support blocks defined in __main__ (#252)

Add local packaging run to workflow tests, with an abbreviated
(blinky-only) unittest and top-level test.

Resolves #251, where blocks defined in `__main__` cause an error when
the HDL server tries to resolve those blocks, because the HDL server is
a separate Python instance and thinks itself is `__main__` (instead of
the original file). The fix is to detect when a library's module is
`__main__` and effectively guess the correct original module name by
using the filename. Might be a better way to do it, but this works, and
is larg [... cut]
closed 2023-05-08 by ducky64