BerkeleyHCI/PolymorphicBlocks #323: PMOS reverse protections

Suke0811, 2024-03-29:
## Pmos

From 1.
[image]

Implementation: 
https://github.com/BerkeleyHCI/PolymorphicBlocks/blob/5bb5b504e6a5da03bbd40ef9a66a422e02ea0a58/electronics_lib/PowerConditioning.py#L267-L305

## PMOS for bi-directional application (Low voltage)

From 2. 
[image]

Implementation: 

https://github.com/BerkeleyHCI/PolymorphicBlocks/blob/5bb5b504e6a5da03bbd40ef9a66a422e02ea0a58/electronics_lib/PowerConditioning.py#L310-L359

Currently, this code gets stuck during the netlist generations
[image]

## Ref
1.  https://components101.com/articles/design-guide-pmos-mosfet-for-reverse-voltage-polarity-protection
2.  https://www.edn.com/reverse-voltage-protection-for-battery-chargers/

Suke0811, 2024-04-01:
- duplicated r1/r2 naming issues now detected with ducky update (5cbf405b4c51cfc1389c8e80515faf2c7b2c650b)

- Updated the charging protection to use schematics
chg and the power_out is manually connected, and the block looks like this?
[image]

```python
    chg_adapter = self.Block(PassiveAdapterVoltageSink())
    setattr(self, '(adapter)chg', chg_adapter)  # hack so the netlister recognizes this as an adapter
    self.connect(self.mp1.source, chg_adapter.src)
    self.connect(self.chg, chg_adapter.dst)
    ```

ducky64, 2024-04-01:
Yeah, the diagram weirdness is from some weirdness in how the underlying layout libraries handles connections between exterior ports.

Suke0811, 2024-04-30:
#328